nxp-mcuxpresso/spsdk #5: Command B2 00 08 21 00 00 00 02 causes 4 bytes skipped

saper, 2020-09-03:
`B2 00 08 21 00 00 00 02` seems to skip 16 bytes, not just 8.

it seems that

https://github.com/NXPmicro/spsdk/blob/fc5d524769f0b7bf047546852eda9bad95bdb383/spsdk/image/commands.py#L765

seems to assume it is always 16 bytes long. 

I have worked it around by adding

```diff
diff --git a/spsdk/image/commands.py b/spsdk/image/commands.py
index 09c5b76..ac50ad7 100644
--- a/spsdk/image/commands.py
+++ b/spsdk/image/commands.py
@@ -762,7 +762,10 @@ class CmdUnlock(CmdUnlockAbstract):
         """
         super().__init__(engine, features)
         self.uid = uid
-        self._header.length = CmdHeader.SIZE + 12
+        if uid > 0:
+            self._header.length = CmdHeader.SIZE + 12
+        else:
+            self._header.length = CmdHeader.SIZE + 4

     def __repr__(self) -> str:
         return "CmdUnlock <{}, {}, {}>".format(
````

but that is probably not a correct fix (it helped me to parse the rest of the CSF, almost).

mstarecek, 2020-09-07:
Hello Saper, many thanks for submitting the issue. 
Yes, you are correct; the UID is mandatory only for some combinations of Engine and Features, not always.

mstarecek, 2020-09-08:
Hello Saper, could you please check pr #6 
After implementing the fix, the CmdUnlock.parse returns the following info: engine=OCOTP, fatures=2 (all off, but SRK-REVOKE)
In this case, there's no need for UID.

closed 2020-09-11 by vhamersky