nxp-mcuxpresso/spsdk #17: tests/mcu_examples/test_rt10xx.py fails under some circumstances in the container

saper, 2021-03-16:
The following tests fail (both for 0.2.1 as well as 1.3) for me when run in Centos 8.2-based container:

========================================================================== short test summary info ==========================================================================
FAILED tests/mcu_examples/test_rt10xx.py::test_sb[rt102x] - AssertionError
FAILED tests/mcu_examples/test_rt10xx.py::test_sb[rt105x] - AssertionError
FAILED tests/mcu_examples/test_rt10xx.py::test_sb[rt106x] - AssertionError
FAILED tests/mcu_examples/test_rt10xx.py::test_sb_multiple_sections[rt105x] - AssertionError
FAILED tests/mcu_examples/test_rt10xx.py::test_sb_multiple_sections[rt106x] - AssertionError

That must be something in the environment; the tests do not fail when run on the RedHat 8.2 machine on in other environments.  The container fails the same way if run under podman on the RedHat 8.2 box or under Kubernetes. Does not matter if pip 21.0.1 (not supported according to the readme - but works outside of the container) or 20.2.4 is installed.

The following files are generated by the build and the hashes inside are indeed different when compared to a clean build:

```
data/rt102x/debug_logs/evkmimxrt1020_iled_blinky_ext_FLASH_unsigned_nopadding.sb.txt.generated
data/rt106x/debug_logs/evkmimxrt1060_iled_blinky_ext_FLASH_unsigned_nopadding.sb.txt.generated
data/rt106x/debug_logs/sb_file_2_sections.sb.txt.generated
data/rt105x/debug_logs/evkbimxrt1050_iled_blinky_ext_FLASH_unsigned_nopadding.sb.txt.generated
data/rt105x/debug_logs/sb_file_2_sections.sb.txt.generated
```

Dockerfile looks looks a bit like this (customer-specific stuff redacted, pip.conf points to the local pip repository on Artifactory).

```
FROM <customer-repository>/centos:8.2.2004
USER root
COPY container_data/local-pip-root-certificate.crt /etc/pki/ca-trust/source/anchors/
COPY container_data/repo/*.repo /etc/yum.repos.d/
RUN /bin/update-ca-trust
RUN dnf module install -y swig:3.0 python36:3.6 rust-toolset:rhel8
RUN dnf install -y gcc-c++ java-11-openjdk-headless python36-devel git util-linux
COPY container_data/pip.conf /etc/
RUN python3 -m pip install --upgrade pip==20.2.4
RUN pip install \
    setuptools==53.0.0 \
    wheel==0.36.2 \
    setuptools_rust==0.11.6 \
    setuptools_scm==5.0.1 \
    pyscaffold==3.2.3
RUN install -d -o 1000 -g 1000 -m 0755 /me
USER 1000
ENV HOME=/me PATH=/me/.local/bin:${PATH}
ENTRYPOINT ["sh"]
CMD ["-i"]
```

mstarecek, 2021-03-16:
Hello Saper, thanks for submitting the issue. Although nor Centos nor RedHat are not officially supported, I'd like to investigate more on the subject. Could you please attach the *.generated files?
Thanks!

saper, 2021-03-16:
I can't seem to be able to attach them here... I'll try to gist them

mstarecek, 2021-03-16:
Ok, if that doesn't work, feel free to send me them directly to: michal.starecek@nxp.com

saper, 2021-03-16:
Here they are: https://gist.github.com/d9a58df35c97aad1cf45afae5ed1ab6f

mstarecek, 2021-03-17:
I'm having a bit of difficulty setting up the docker.
However, I was able to reproduce the issue on my side (to a certain extent). Issues indicated by *.generated files were caused by missing timezone information when creating timestamps in tests. It would seem that you were the first one, who took SPSDK outside central Europe :D 
Could you please try out branch  issue-17-test_rt-timezone just to make sure there are no other (container-related) issues? 
Many thanks!

saper, 2021-03-17:
Thank you, that fixes the problem!

The funny part is that - I am in Central Europe and we use the same timezone as you do. The container has no `$TZ` set which means UTC but many programs have problem with it. 

I was running pytest with `env LANG=C TZ=UTC pytest` since I suspected it might be either locale or the timezone.

But whatever it was, the change works!

closed 2021-03-17 by mstarecek