nxp-mcuxpresso/spsdk #22: Report progress in McuBoot.receive_sb_file

robin-nitrokey, 2022-01-11:
It would be useful if the [`McuBoot.receive_sb_file`](https://spsdk.readthedocs.io/en/latest/api/mboot.html#spsdk.mboot.mcuboot.McuBoot.receive_sb_file) method would provide a way to check the progress of the operation.

mstarecek, 2022-01-11:
Hi, this sure sounds interesting.
On top of my head, I can think of having an optional parameter for McuBoot, a callable that expects two parameters: current iteration, total iterations... This will allow users to do pretty much whatever they want.
Please, let me know your thoughts.

robin-nitrokey, 2022-01-12:
Having a callback as you described sounds good to me!  I think it would me more intuitive to have it as an optional argument for `receive_sb_file` (and maybe other applicable operations), not for the `McuBoot` instance, but both should work for us.

mstarecek, 2022-01-12:
I discussed the topic with my colleagues, and we decided to implement both options: setting the callback during McuBoot instantiation and also as an optional argument in all functions that transfer data blocks (receive_sb_file, read_memory, load_image etc.) Callback defined in function call takes precedence over callback defined in "constructor".

mstarecek, 2022-02-04:
As of release 1.6.0, the [McuBoot.receive_sb_file](https://spsdk.readthedocs.io/en/latest/api/mboot.html#spsdk.mboot.mcuboot.McuBoot.receive_sb_file) (amongst other methods that transfer data) accepts a callback method for notifying called about the progress of data transfer. The first argument is the amount of data already transferred, the second one is the total amount to transfer.

closed 2022-02-04 by mstarecek