nxp-mcuxpresso/spsdk #23: Provide error details in McuBoot.receive_sb_file

robin-nitrokey, 2022-01-11:
Currently, [McuBoot.receive_sb_file](https://spsdk.readthedocs.io/en/latest/api/mboot.html#spsdk.mboot.mcuboot.McuBoot.receive_sb_file) returns a boolean indicating the operation result.  It would be useful to access more details about the error if the operation fails.

mstarecek, 2022-01-11:
Hi, in case of a failure you can refer to [McuBoot.status_code](https://spsdk.readthedocs.io/en/latest/api/mboot.html#spsdk.mboot.mcuboot.McuBoot.status_code) property for more details. You may get a stringified error message for the status_code using `spsdk.mboot.StatusCode.desc(status_code)`
In the next release, we plan to add a new property "status_string" that will return a status string message directly.

Another way is to use McuBoot with `cmd_exception=True`. With this setting McuBoot will throw an exception based on `spsdk.SPSDKError` in case of an error.

robin-nitrokey, 2022-01-12:
I somehow missed that, thanks for the pointers!

closed 2022-01-12 by robin-nitrokey