nxp-mcuxpresso/spsdk #25: libusbsio: Invalid format string if connection fails

robin-nitrokey, 2022-01-18:
(This is an issue in the libusbsio dependency, but I’m reporting it here because I did not find an issue tracker for it.)

----

If the connection to a device fails, for example because of a permission problem, an invalid format string is used in `libusbsio.py`:

```
    ​2             self._h = self._sio._HIDAPI_DeviceOpen(path)
    ​1             self.logger.info("HID device %d is now open" % self._h)
 ​1323             return self._h != 0
```

In this case, `self.h` is `None` and cannot be formatted with `%d`.

```
Traceback (most recent call last):
  File "/home/jens/.local/lib/python3.10/site-packages/spsdk/mboot/interfaces/usb.py", line 152, in open
    self.device.Open(self.path)
  File "/home/jens/.local/lib/python3.10/site-packages/libusbsio/libusbsio.py", line 1322, in Open
    self.logger.info("HID device %d is now open" % self._h)
TypeError: %d format: a real number is required, not NoneType
```

Full error message here: https://github.com/Nitrokey/pynitrokey/issues/153

mstarecek, 2022-01-25:
Hi, libusbsio is not under our control, but we're in touch with the owner who promised an update soon.

robin-nitrokey, 2022-01-27:
Thanks for forwarding the issue!

closed 2022-01-27 by robin-nitrokey