nxp-mcuxpresso/spsdk #27: click 7.1 is required

robin-nitrokey, 2022-01-21:
The `requirements.txt` file lists 7.0 as the minimum `click` version but the project uses the [`no_args_is_help`](https://click.palletsprojects.com/en/7.x/api/#click.Command) argument for `click.Command`/`click.command`/`click.group` that was added in `click` 7.1.

robin-nitrokey, 2022-01-21:
On a related note, it would be useful if library users could disable the CLI and the click dependency, maybe using an extra, to avoid version conflicts and unnecessary dependencies.

mstarecek, 2022-01-25:
Hi, thanks for bringing this up. It will be fixed in the upcoming release, scheduled for early next week. Disabling CLI sure sounds interesting, but it won't happen in this release, perhaps we could cycle back to the topic in the next iteration.

mstarecek, 2022-02-04:
The requirements file was updated. As of right now, we don't plan to disable CLI (or split SPSDK into API and CLI packages)

closed 2022-02-04 by mstarecek