nxp-mcuxpresso/spsdk #28: cryptography 3.4 is required

robin-nitrokey, 2022-01-21:
spsdk requires cryptography >=3.3,<3.4.4, but running with 3.3 causes this error for me:

```
Traceback (most recent call last):
  File "/home/robin/reps/pynitrokey/./venv/bin/nitropy", line 5, in <module>
    from pynitrokey.cli import nitropy
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/pynitrokey/cli/__init__.py", line 20, in <module>
    from pynitrokey.cli.nk3 import nk3
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/pynitrokey/cli/nk3/__init__.py", line 21, in <module>
    from pynitrokey.nk3 import list as list_nk3
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/pynitrokey/nk3/__init__.py", line 13, in <module>
    from .bootloader import Nitrokey3Bootloader
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/pynitrokey/nk3/bootloader.py", line 15, in <module>
    from spsdk.mboot import McuBoot, StatusCode
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/mboot/__init__.py", line 20, in <module>
    from .mcuboot import McuBoot
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/mboot/mcuboot.py", line 37, in <module>
    from .memories import ExtMemId, ExtMemRegion, FlashRegion, MemoryRegion, RamRegion
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/mboot/memories.py", line 13, in <module>
    from spsdk.image.misc import size_fmt
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/image/__init__.py", line 12, in <module>
    from .commands import (
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/image/commands.py", line 19, in <module>
    from spsdk.crypto import Certificate, Encoding
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/crypto/__init__.py", line 66, in <module>
    from .loaders import (
  File "/home/robin/reps/pynitrokey/venv/lib/python3.9/site-packages/spsdk/crypto/loaders.py", line 11, in <module>
    from cryptography.hazmat._types import _PRIVATE_KEY_TYPES as PrivateKey
ModuleNotFoundError: No module named 'cryptography.hazmat._types'
```

Using cryptography 3.4 fixes the problem.

mstarecek, 2022-01-25:
Hi, thanks for bringing this up. It will be fixed in the upcoming release, scheduled for early next week

mstarecek, 2022-02-04:
The minimal required version of cryptography was bumped to version 3.4.4. Since this version, the cryptography package contains py.typed

closed 2022-02-04 by mstarecek