nxp-mcuxpresso/spsdk #29: Receiving SB files is slow

robin-nitrokey, 2022-01-24:
From my experience, receiving SB files with spsdk is much slower than with for example mboot or other implementations.  This affects both calling the `McuBoot.receive_sb_file` function as well as executing `spsdk blhost receive-sb-file`.  On Linux, the time difference is ~ 3 s vs. 1 min.  On Windows, the time difference is ~ 10 s vs. 12 min.

mstarecek, 2022-01-25:
Hello, the timing discrepancy is coming from importing various modules (this heavily depends on your computer setup). Running `spsdk blhost` eventually imports everything that is in the spsdk (although the majority is not used at all).
For running CLI commands I strongly recommend using the CLI tool directly (e.g.: `blhost receive-sb-file`) The `spsdk` wrapper is there only as a showcase of all available commands and for people who can't use the tools directly, because they have tools with conflicting names in PATH.
Unfortunately, at the moment we don't have enough time/resources to investigate the timing bottle-necks. Hopefully we'll get to it soon.

robin-nitrokey, 2022-01-27:
Thans for the explanation!  Using `blhost` instead of `spsdk blhost` does not make a big difference for me, at least in the receive-sb2-file case (1m1.484s vs 1m2.342s).  After profiling the `receive_sb_file` method, I think reading the response in `libusbsio` is causing the slow speed:

```
ncalls  tottime  percall  cumtime  percall filename:lineno(function)
  5435   55.821    0.010   55.853    0.010 libusbsio.py:1365(Read)
```

mstarecek, 2022-01-27:
The long Read depends on the length of the SB file and chosen signature/encryption schema. During the receive-sb-file operation, the target verifies the signature and may re-encrypt the SB file itself. Read method is basically waiting for the ACK from the target. It would be helpful for me to know what device do you use, type of singing key (RSA/ECC), and a basic boot setting.
It's understandable if you don't want to disclose such info here, feel free to send an email.

robin-nitrokey, 2022-01-28:
That makes sense, still `mboot` and `lpc55-host` are much faster.  Maybe they are just not waiting for the ACK?

Anyway, my main reason for opening this issue was the very slow execution time on Windows.  1 minute transfer time is okay, 12 minutes is rather annoying.  I’ll try to run the profiling on Windows too, maybe that can give some insight.

robin-nitrokey, 2022-01-29:
On Windows, the Write calls are causing the slow speed:

```
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
     5404  593.373    0.110  593.406    0.110 libusbsio.py:1339(Write)
     5405   84.860    0.016   84.875    0.016 libusbsio.py:1364(Read)
```

mstarecek, 2022-01-31:
This is very interesting. We'll investigate more because this is odd.
Libusbsio uses HIDAPI, same as `mboot` (and presumably `lpc55-host`. according to its Cargo.toml and the hidapi crate docs) 
There's only a slight modification to hidapi in libusbsio, so it will work also for other NXP devices. But this change should not yield such timing differences.
Many thanks for bringing this up.

robin-nitrokey, 2022-01-31:
I’ve had another look at the profiling data from Linux:

```
   ncalls  tottime  percall  cumtime  percall filename:lineno(function)
        1    0.000    0.000   62.011   62.011 mcuboot.py:537(receive_sb_file)
        1    0.065    0.065   62.007   62.007 mcuboot.py:166(_send_data)
     5434    0.293    0.000   61.926    0.011 usb.py:176(write)
    10870    0.200    0.000   60.700    0.006 libusbsio.py:1301(wrapper)
     5435   55.875    0.010   55.910    0.010 libusbsio.py:1365(Read)
     5434    4.493    0.001    4.564    0.001 libusbsio.py:1340(Write)
```

My first impression was that reading the command response in `RawHid.read` was the slow part.  But it turns out that the `RawHid.write` method also calls the `libusbsio.Read` method (“try to read a begging of the ABORT_FRAME”) and is actually causing the slowdown:  As the data is split into chunks, `RawHid` tries to read a response from the device before writing each chunk – in total 5434 times ([code](https://github.com/NXPmicro/spsdk/blob/master/spsdk/mboot/interfaces/usb.py#L200-L209)).  As the timeout is set to 10 ms and there is no  abort frame to read, this results in 5434 * 10 ms = 54.34 unnecessary idle time. Is this read necessary before every chunk?  Could it be performed every 10, 100 or 1000 chunks instead?  Could the timeout be reduced?

mstarecek, 2022-02-02:
Many thanks for your investigation.
I totally forgot about the ABORT_FRAME. As you might probably guess from the nature of the code (the clunky passing of the `allow_abort` flag), this feature was added only as an afterthought for different devices. This also might be a reason why `mboot` and `lpc55-host` projects didn't pick upon this.
I did some measurements on my end and I can confirm it's the anticipation of the ABORT_FRAME that is causing such poor performance. (in my case for 400kB SB file the time went down from 1m45s to 5s!!!
I'll investigate more on how the change in checking for ABORT_FRAME would affect other NXP devices. Meanwhile, we plan to add an option into `receive-sb-file` API/CLI to disable the check.

mstarecek, 2022-02-04:
The [receive-sb-file](https://spsdk.readthedocs.io/en/latest/api/mboot.html#spsdk.mboot.mcuboot.McuBoot.receive_sb_file) and corresponding [blhost sub-command](https://spsdk.readthedocs.io/en/latest/apps/blhost.html#blhost-receive-sb-file) now by default skip the ABORT_FRAME checking resulting in a huge performance boost when using USB interface. Error checking can be enabled via the `check-errors` parameter/flag.

closed 2022-02-04 by mstarecek

robin-nitrokey, 2022-02-04:
Thanks for the fix!  Does this also fix the slow writes on Windows?

robin-nitrokey, 2022-02-04:
Also, I understand that the abort frame feature is only relevant for some devices.  Can you add a list of these devices?  Is the LPC55 one of them?

mstarecek, 2022-02-07:
This fix is common for all OSs. Writes on Windows will always be slower compared to Linux/Mac, but there's not really much we can do about it :(

The ABORT_FRAME is relevant for all devices. But in LPC55 devices the error code user gets is correct, regardless of whether they use the check-errors flag or not. In case of some RT5xx/6xx devices, users may get a different error code when they don't enable the ABORT_FRAME checker (it's very difficult to pinpoint which exact devices are affected, as this depends on silicon revision). 

In summary: Not checking for ABORT_FRAME makes the communication faster. However, in case of a failure, users may get an incorrect (misleading) error code. Therefore, in case of an error, it's recommended to re-run the receive-sb-file operation to make sure the error code represents the true root cause.