nxp-mcuxpresso/spsdk #31: Support for parsing master boot images

robin-nitrokey, 2022-02-01:
` spsdk.image.mbimg.MasterBootImage.parse` is currently not implemented.  I think it would make sense to either remove the method altogether until it is implemented, or at least mention in the documentation that it does not work.  Do you have plans for implementing this function?

mstarecek, 2022-02-01:
Hi, I always thought that the docstring contains such a message. Guess we need to double-check our pylint and pydocstyle settings :)
We do intend to implement this feature. However, currently, it has low priority. For now, we'll just add NotImplemented info into the method's docstring.

robin-nitrokey, 2022-02-01:
Okay, thank you!

mstarecek, 2022-02-04:
Docstring (and generated docs) now properly state if a method is not implemented.

closed 2022-02-04 by mstarecek