nxp-mcuxpresso/spsdk #34: Document versioning policy

robin-nitrokey, 2022-02-07:
The v1.6.0 release broke compatibility with the previous releases because the `spsdk.sbfile.images` module was removed.  To avoid such changes affecting other projects, it would be helpful to know your versioning policy.  Which library versions can be assumed to be compatible?

mstarecek, 2022-02-14:
The SPSDK project is still far from being a stable product (in PyPI it's marked as alpha).
I know the `1.` in version is very misleading, but this is a corporate "can of worms" I don't want to open :)
Considering the version being `major.minor.patch` we can't promise everything will be compatible with a "minor" release (1.5->1.6). What we want to maintain is backward compatibility in "patch" releases.
We'll add this kind of disclaimer either directly in README or in generated docs.

marekvi95, 2022-09-27:
I am closing this issue, the versioning policy was already described by @mstarecek.

closed 2022-09-27 by marekvi95