nxp-mcuxpresso/spsdk #39: SPSDK 1.6.0 and 1.5.0 fails after MarkupSafe 2.1.0 release

ivan2angelov, 2022-02-18:
SPSDK requirements.txt needs:
jinja2>=2.11,<3
jinja2 happily pulls Markupsafe larger or equal to 2.0, see:
https://github.com/pallets/jinja/blob/main/setup.py

Very recently https://pypi.org/project/MarkupSafe/2.1.0/ was released (link https://markupsafe.palletsprojects.com/en/2.1.x/changes/):
"Remove soft_unicode, which was previously deprecated. Use soft_str instead. [#261](https://github.com/pallets/markupsafe/pull/261)"
As a result when SPSDK uses jinja2 there is an exception due to missing import:
ImportError: cannot import name 'soft_unicode' from markupsafe

Suggestion: could you fix jinja2 version in SPSDK to a version that will guarantee operation of the SPSDK?

Gargy007, 2022-02-18:
Hello,
 thanks for good point, you nicely confirm just our findings from today morning. The issue will be solved in release SPSDK v1.6.1 which will be release this month.

Gargy007, 2022-03-16:
> @Gargy007 <https://github.com/Gargy007> I can see 1.6.1 release and 1.6.2
> release but I am not sure if this issue was fixed. Can you check if a fix
> was released?

Hi, it should be. I moved the jinja2 to Hitler version.

Gargy

marekvi95, 2022-12-02:
Closing the issue, Jijna version has been bumped to 3.0. Upper boundary is now being handled by CI system.

closed 2022-12-02 by marekvi95