nxp-mcuxpresso/spsdk #48: Fails to parse BD file with version_check

robin-nitrokey, 2022-08-03:
The following `commands.bd` file as suggested in [this thread in the NXP forum](https://community.nxp.com/t5/LPC-Microcontrollers/LPC55S69-Secure-Boot-Failing/m-p/1017769#M39800) is rejected by `spsdk`’s BD parser and therefore also by its `elftosb` and `nxpimage` apps while it works with the legacy `elftosb` implementation.

`commands.bd`:
```
options {
flags = 0x8; // 0x8 encrypted + signed, 0x4 encrypted
buildNumber = 0x1;
productVersion = "1.00.00";
componentVersion = "1.00.00";
secureBinaryVersion = "2.1";
}
sources {
inputFile = extern(0);
}
section (0) {
version_check sec 0x2;
version_check nsec 2;
erase 0x0..0x40000;
load inputFile > 0x0;
}
```

Example code:
```py
from spsdk.sbfile.sb2.sly_bd_parser import BDParser

parser = BDParser()
with open("commands.bd") as f:
    bd = parser.parse(f.read(), extern=["1"])
print(bd)
```

Error:
```
spsdk.exceptions.SPSDKError: SPSDK: bdcompiler:12:1: error

version_check sec 0x2;
^
```

marekvi95, 2022-08-10:
Hello, thank you for reporting this issue. I have already implemented "version_check" SB command in our internal repository.  Once I get the approval, I will push it to GitHub.

dvzrv, 2022-09-24:
@marekvi95 so... is this included in 1.7.1 since that has been released after your statement?

daringer, 2022-09-25:
nope, for me based on first tests the error still exists with 1.7.1. 
Can you confirm?

marekvi95, 2022-09-25:
Hi @dvzrv and @daringer. I am sorry this fix was not included in 1.7.1 release, we planned to include this fix in 1.7.2 version. The previous PR was created from wrong branch. A moment ago I have created new PR with cherry-picked commit that adds support for version_check command in SB 2.1. It will be merged into master branch. I am also sorry that it took so long.

commit 6fc1c7b4 (2022-09-26): Add support for version_check command, fixes #48 (#50)
closed 2022-09-26 by mwsk