nxp-mcuxpresso/spsdk #53: Support *BSD

robin-nitrokey, 2023-04-04:
`libusbsio` does not include a binary for *BSD systems so `spsdk` cannot be used on these systems.  Please consider adding *BSD support.

A developer using our software was able to manually patch `libusbsio` to compile on FreeBSD.  He has the following suggestions/questions:  Can you

> - put their sources to github or some other repository it is possible to provide patches for?
> - clarify if we really need `ex_info` in `HIDAPI_ENUM_T` for LPC55 (this seems to be unused by the Python part and SPSDK)
> - Do we really need `timeout_ms` in `HIDAPI_DeviceWrite` and if yes, maybe the `hid_write_timeout` should have been moved inside of that function (and probably rewritten, since it looks pretty strange to me).

mstarecek, 2023-04-04:
Hello Robin, `libusbsio` is its own project and we are simply using it as a dependency. 
We will forward this issue to the project maintainer. However, we can't guarantee that it will be resolved.

saper, 2023-04-04:
Thank you @mstarecek - OP here. Yes, the main problem is there is no place to discuss eventual changes to `libusbsio`. I think it could easily support upstream, unchanged https://github.com/signal11/hidapi. If I read the code correctly, modifications to hidapi are not needed for the SPSDK at all. But I understand this library needs to support probably more than we can see here. Maybe this project could be brought over to GitHub or somewhere else in the open?