nxp-mcuxpresso/spsdk #60: *.c files in spsdk/data/cpu_data required when packaging this?

johanneskastl, 2023-05-29:
Hi all,

during packaging this for openSUSE (as a dependency of pynitrokey) I noticed that there are some `*.c` files (and some `*.bin` ones) in `spsdk/data/cpu_data`.

Are those *.c files required for normal operation?
Are the *.bin files required?

Should the *.c files be compiled (which would make this an arch-dependent package, not a noarch one).

Thanks in advance!

marekvi95, 2023-05-30:
Hi Johannes,

These files should not be part of the package. We will move them to example section in the next release. The purpose of these files is to demonstrate a way how to do a HAB log audit in case something went wrong during HAB boot.

Marek

johanneskastl, 2023-05-31:
Thanks a lot @marekvi95 for the fast reply. I will drop them from the package.

closed 2023-05-31 by johanneskastl