nxp-mcuxpresso/spsdk #62: In LPC55Sxx Secure Boot documentation, describe how to set ROTKH in CMPA

cmcqueen, 2024-02-04:
I'm working through the [LPC55Sxx Secure Boot](https://spsdk.readthedocs.io/en/2.0.1/examples/lpc55sxx_secure_boot.html) page in the SPSDK documentation. It describes how to prepare and write the data for the CMPA (Customer Manufacturing Programming Area) and CFPA (Customer in-field Programming Area) in the PFR (Protected Flash Region).

The documentation doesn't say anything about setting the ROTKH in the CMPA. But if I understand the concepts, isn't this a crucial step to making the secure boot actually secure? Is this missing from the documentation?

marekvi95, 2024-02-05:
Hello Craig,

Sure, ROTKH in CMPA must be set to get the secure boot working. There a three ways to accomplish that. You might set ROTKH in the PFR configuration directly or you might provide a path to the certificate block or master boot image configuration with (-e or --rot-config) https://spsdk.readthedocs.io/en/latest/apps/pfr.html#cmdoption-pfr-generate-binary-e

The last way is to use the --secret-file option, where you can specify paths to secret files (keys or certificates) that will be used for calculating the ROTKH value.

I agree with you that it is not clear in the documentation, we will try to improve it in the next version. Thanks for the feedback.

Marek

mwsk, 2024-03-28:
Next release (2.2.0) is planned for 7th June. There will be update in example as following: 
"ROTKH in CMPA must be set to get the secure boot working. There a three ways to accomplish that. You might set ROTKH in the PFR configuration directly or you might provide a path to the certificate block or master boot image configuration with (-e or --rot-config) option.

The last way is to use the --secret-file option, where you can specify paths to secret files (keys or certificates) that will be used for calculating the ROTKH value."

mwsk, 2024-06-16:
https://spsdk.readthedocs.io/en/latest/examples/lpc55sxx_secure_boot/lpc55sxx_secure_boot.html

mwsk, 2024-06-16:
ROTKH in CMPA must be set to get the secure boot working. There a three ways to accomplish that. You might set ROTKH in the PFR configuration directly or you might provide a path to the certificate block or master boot image configuration with (-e or –rot-config) option.

The last way is to use the –secret-file option, where you can specify paths to secret files (keys or certificates) that will be used for calculating the ROTKH value.

closed 2024-06-16 by mwsk