nxp-mcuxpresso/spsdk #66: 2.1.1: Failing tests due to missing files

dvzrv, 2024-03-31:
Hi! :wave: 

When trying to upgrade the package on Arch Linux to 2.1.1 I ran into quite a few failing tests due to missing files.

```
=========================== short test summary info ============================
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp256r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp384r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp521r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-rsa2048]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-rsa4096]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA256-secp384r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA256-secp521r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-secp256r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-secp521r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-rsa2048]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-rsa4096]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA512-secp256r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA512-secp384r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA512-rsa2048]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA512-rsa4096]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-secp256r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-secp384r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-secp521r1]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-rsa2048]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-rsa4096]
FAILED tests/nxpimage/test_nxpimage_sb31.py::test_nxpimage_sb31_kaypair_not_matching
= 21 failed, 2436 passed, 20 skipped, 1 xpassed, 114 warnings in 76.08s (0:01:16) =
```

Build/check logs:
[python-spsdk-2.1.1-1-x86_64-build.log](https://github.com/nxp-mcuxpresso/spsdk/files/14816458/python-spsdk-2.1.1-1-x86_64-build.log)
[python-spsdk-2.1.1-1-x86_64-check.log](https://github.com/nxp-mcuxpresso/spsdk/files/14816456/python-spsdk-2.1.1-1-x86_64-check.log)

I'm already skipping some doc tests as they are not relevant for our use-case, but the above failures appear to be due to missing files.

When looking at e.g.

```
>           raise SPSDKError(f"Can't load configuration file: {str(exc)}") from exc
E           spsdk.exceptions.SPSDKError: SPSDK: Can't load configuration file: SPSDK: Path '/build/python-spsdk/src/spsdk-2.1.1/tests/nxpimage/data/workspace/cfgs/lpc55s3x/sb3_256_256_keys_dont_match.yaml' not found
```

```bash
ls -lah /var/lib/archbuild/extra-testing-x86_64/dave-5/build/python-spsdk/src/spsdk-2.1.1/tests/nxpimage/data/workspace/cfgs/lpc55s3x/
total 128K
drwxr-xr-x 1 dave dave 1.2K Mar 31 17:09 .
drwxr-xr-x 1 dave dave  130 Mar 31 17:09 ..
-rw-r--r-- 1 dave dave 2.1K Mar 31 17:09 mb_ext_xip_crc.yaml
-rw-r--r-- 1 dave dave 1.6K Mar 31 17:09 mb_ext_xip_crc_s19.yaml
-rw-r--r-- 1 dave dave 2.7K Mar 31 17:09 mb_ext_xip_signed.yaml
-rw-r--r-- 1 dave dave 2.1K Mar 31 17:09 mb_ram_crc.yaml
-rw-r--r-- 1 dave dave 2.2K Mar 31 17:09 mb_ram_crc_version.yaml
-rw-r--r-- 1 dave dave 3.1K Mar 31 17:09 mb_xip_256_none.yaml
-rw-r--r-- 1 dave dave 2.6K Mar 31 17:09 mb_xip_256_none_no_tz.yaml
-rw-r--r-- 1 dave dave 2.6K Mar 31 17:09 mb_xip_384_256.yaml
-rw-r--r-- 1 dave dave 2.9K Mar 31 17:09 mb_xip_384_256_cert.yaml
-rw-r--r-- 1 dave dave 1.5K Mar 31 17:09 mb_xip_384_256_cert_invalid.json
-rw-r--r-- 1 dave dave 2.6K Mar 31 17:09 mb_xip_384_384.yaml
-rw-r--r-- 1 dave dave 3.0K Mar 31 17:09 mb_xip_384_384_cert.yaml
-rw-r--r-- 1 dave dave 2.1K Mar 31 17:09 mb_xip_crc.yaml
-rw-r--r-- 1 dave dave 3.8K Mar 31 17:09 sb3_256_256.yaml
-rw-r--r-- 1 dave dave 4.4K Mar 31 17:09 sb3_256_256_cert.yaml
-rw-r--r-- 1 dave dave 5.4K Mar 31 17:09 sb3_256_256_notime.yaml
-rw-r--r-- 1 dave dave 3.9K Mar 31 17:09 sb3_256_none.yaml
-rw-r--r-- 1 dave dave 3.6K Mar 31 17:09 sb3_256_none_ernad.yaml
-rw-r--r-- 1 dave dave 3.8K Mar 31 17:09 sb3_384_256.yaml
-rw-r--r-- 1 dave dave 4.2K Mar 31 17:09 sb3_384_256_cert.yaml
-rw-r--r-- 1 dave dave 3.8K Mar 31 17:09 sb3_384_256_fixed_timestamp.yaml
-rw-r--r-- 1 dave dave 4.1K Mar 31 17:09 sb3_384_256_unencrypted.yaml
-rw-r--r-- 1 dave dave 3.5K Mar 31 17:09 sb3_384_384.yaml
-rw-r--r-- 1 dave dave 3.9K Mar 31 17:09 sb3_384_384_cert.yaml
-rw-r--r-- 1 dave dave 3.9K Mar 31 17:09 sb3_384_none.yaml
-rw-r--r-- 1 dave dave  11K Mar 31 17:09 sb3_test_384_384_unencrypted.yaml
```

I will disable all tests in question.

mstarecek, 2024-04-04:
This issue was resolved in #70

jjakob, 2024-04-04:
Now there are assertion errors
```
___________________________ test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-secp521r1] ____________________________

cli_runner = <tests.cli_runner.CliRunner object at 0x7eff3abf9650>
data_dir = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data'
tmpdir = local('/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/temp/pytest-of-portage/pytest-0/test_nxpcrypto_create_signatur17')
key_type = 'secp521r1', algorithm = <EnumHashAlgorithm.MD5: SpsdkEnumMember(tag=4, label='md5', description='MD5')>

<snipped code>

>       assert pub.verify_signature(signature, load_binary(input_file), **extra_params)
E       assert False
E        +  where False = <bound method PublicKeyEcc.verify_signature of ECC EccCurve.SECP521R1 Public Key>(b'0\x81\x87\x02B\x00\x80\xfaC\xf6P\xcfa\x8aX:\xa9\xf6\xc2\x1d\xb8\xbc\xd7y\x12\x1cl\x13E.D\xa7W"\x89\xcc\xcfN\xfeS*\x96\xe1\xbbs\x17\xa7zX\x19>\xb1\xbf\x92\xa3Y\xef\x906\x8c\x18=\xab\xd7Y\xd9>\x8f\xd7p\x85\x02Ai\xdb\x19F$\xe9#\xfbT\x91\x06\ntV6\x93\xb5bt\x9a\x06\x05v\x88\xcd4\xbbY\x19\xfd\x9a\xb9Xpo\x8d\xedr`\x1b"\x19\x1b\x04\x05\x02-"\x13\xb8\x98\xa7\xda\xbc\xc3\xdf\x14\x8fO0\xbdJ$Z\x96', b'2\x1f7$\xb5\xd1\x0b\xb3\xe6\xe2p;\xd5\x19\x13\x1e\xefM\xe5\x89\xb2\x88\xd5\xf0\xd39\xf2\x99\xffu?/', **{'algorithm': <EnumHashAlgorithm.MD5: SpsdkEnumMember(tag=4, label='md5', description='MD5')>})
E        +    where <bound method PublicKeyEcc.verify_signature of ECC EccCurve.SECP521R1 Public Key> = ECC EccCurve.SECP521R1 Public Key.verify_signature
E        +    and   b'2\x1f7$\xb5\xd1\x0b\xb3\xe6\xe2p;\xd5\x19\x13\x1e\xefM\xe5\x89\xb2\x88\xd5\xf0\xd39\xf2\x99\xffu?/' = load_binary('/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/data_to_sign.bin')

algorithm  = <EnumHashAlgorithm.MD5: SpsdkEnumMember(tag=4, label='md5', description='MD5')>
cli_runner = <tests.cli_runner.CliRunner object at 0x7eff3abf9650>
cmd        = ('signature create -k '
 '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/signature_tool/secp521r1.pem '
 '-i '
 '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/data_to_sign.bin '
 '-o '
 '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/temp/pytest-of-portage/pytest-0/test_nxpcrypto_create_signatur17/signature.bin
'
 '--algorithm md5')
data_dir   = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data'
extra_params = {'algorithm': <EnumHashAlgorithm.MD5: SpsdkEnumMember(tag=4, label='md5', description='MD5')>}
input_file = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/data_to_sign.bin'
key_type   = 'secp521r1'
output_file = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/temp/pytest-of-portage/pytest-0/test_nxpcrypto_create_signatur17/signature.bin'
priv_key   = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/signature_tool/secp521r1.pem'
pub        = ECC EccCurve.SECP521R1 Public Key
pub_key    = '/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/work/spsdk-a99838f79a9249264f881b63857df9235ec96f4d/tests/nxpcrypto/data/signature_tool/secp521r1.pub'
signature  = (b'0\x81\x87\x02B\x00\x80\xfaC\xf6P\xcfa\x8aX:\xa9\xf6\xc2\x1d\xb8\xbc\xd7y'
 b'\x12\x1cl\x13E.D\xa7W"\x89\xcc\xcfN\xfeS*\x96\xe1\xbbs\x17\xa7zX\x19>\xb1'
 b'\xbf\x92\xa3Y\xef\x906\x8c\x18=\xab\xd7Y\xd9>\x8f\xd7p\x85\x02Ai\xdb\x19'
 b'F$\xe9#\xfbT\x91\x06\ntV6\x93\xb5bt\x9a\x06\x05v\x88\xcd4\xbbY\x19\xfd\x9a'
 b'\xb9Xpo\x8d\xedr`\x1b"\x19\x1b\x04\x05\x02-"\x13\xb8\x98\xa7\xda\xbc\xc3'
 b'\xdf\x14\x8fO0\xbdJ$Z\x96')
tmpdir     = local('/var/tmp/portage/dev-python/spsdk-2.1.1_p20240404/temp/pytest-of-portage/pytest-0/test_nxpcrypto_create_signatur17')
```

```
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp256r1] - AssertionError: assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp384r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-secp521r1] - AssertionError: assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-rsa2048] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA1-rsa4096] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA256-secp384r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA256-secp521r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-secp256r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-secp521r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.SHA384-rsa2048] - assert False
[... cut ...]
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-secp521r1] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-rsa2048] - assert False
FAILED tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm[EnumHashAlgorithm.MD5-rsa4096] - assert False
```
 It also prints lots of these warnings
```
WARNING  spsdk.apps.nxpcrypto:nxpcrypto.py:500 Only SM3 hash algorithm is supported for OSCCA
```

mstarecek, 2024-04-04:
This is weird... What OS do you use? Because our internal tests on Win10, 11 Mac 13, and Ubuntu 22, 24 are passing just fine...

jjakob, 2024-04-04:
> This is weird... What OS do you use? Because our internal tests on Win10, 11 Mac 13, and Ubuntu 22, 24 are passing just fine...

Gentoo

jjakob, 2024-04-04:
Here's the complete [build.log](https://github.com/nxp-mcuxpresso/spsdk/files/14876285/build.log)

mstarecek, 2024-04-04:
I'm not completely sure what has happened there, but in the log I see that tests are failing only for unusual (almost bizarre) combinations of key and hash. For example: SECP256 has passed using SHA256 but failed for any other. RSA has passed with SHA256 but failed for any other. SECP384 passed only with SHA384... 
To me, it seems that on Gentoo either OpenSSL or cryptography (the library we use) takes only "valid" or "usual" combinations and rejects everything else...

jjakob, 2024-04-05:
If it helps, these are the versions:
openssl 3.0.13
cryptography 42.0.5

Do you think it's not critical? I can mark them xfail?

mstarecek, 2024-04-05:
There are only few combinations that are mandatory. I'll split the test scope into "mandatory" and "optional" and mark the "optional" ones with `xfail`. In that case if the test passes... good, if it fails, well... not a big deal :)

mstarecek, 2024-04-05:
I don't have access to a Gentoo machine, please try to run the test from branch 66-211-failing-tests-due-to-missing-files
If it works for you, we can then merge PR #71 and hopefully close this ticket as well :)
Thanks

jjakob, 2024-04-05:
Tests are successful now, here's the summary:

```
===================================================== short test summary info =====================================================
SKIPPED [1] tests/crypto/test_oscca.py:10: could not import 'gmssl': No module named 'gmssl'
SKIPPED [1] tests/dk6/test_dk6_tools.py:16: could not import 'pyftdi': No module named 'pyftdi'
SKIPPED [1] tests/crypto/test_sign_provider.py:68: Install OSCCA dependency with pip install spsdk[oscca]
SKIPPED [1] tests/crypto/test_sign_provider.py:92: Install OSCCA dependency with pip install spsdk[oscca]
SKIPPED [1] tests/image/images/test_bootimg_api.py:20: unconditional skip
SKIPPED [1] tests/image/segments/test_csf_api.py:26: unconditional skip
SKIPPED [1] tests/image/segments/test_csf_api.py:31: unconditional skip
SKIPPED [1] tests/mcu_examples/test_rt10xx.py:761: Unsupported configuration
SKIPPED [1] tests/mcu_examples/test_rt10xx.py:854: Unsupported configuration
SKIPPED [3] tests/mcu_examples/test_rt10xx.py:1069: Take a look into this
SKIPPED [1] tests/nxpcrypto/test_nxpcrypto.py:470: OSCCA support is not installed
SKIPPED [4] tests/nxpcrypto/test_nxpcrypto.py:486: OSCCA support is not installed
SKIPPED [2] tests/nxpcrypto/test_nxpcrypto.py:638: OSCCA support is not installed
SKIPPED [1] tests/nxpimage/test_nxpimage_ahab.py:106: Install OSCCA dependency with pip install spsdk[oscca]
SKIPPED [1] tests/nxpimage/test_nxpimage_ahab.py:206: Install OSCCA dependency with pip install spsdk[oscca]
XFAIL tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm_optional[secp256r1-algorithms0] - Some Linux distributions allows only certain combinations of key-hash
XFAIL tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm_optional[secp384r1-algorithms1] - Some Linux distributions allows only certain combinations of key-hash
XFAIL tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm_optional[secp521r1-algorithms2] - Some Linux distributions allows only certain combinations of key-hash
XFAIL tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm_optional[rsa2048-algorithms3] - Some Linux distributions allows only certain combinations of key-hash
XFAIL tests/nxpcrypto/test_nxpcrypto.py::test_nxpcrypto_create_signature_algorithm_optional[rsa4096-algorithms4] - Some Linux distributions allows only certain combinations of key-hash
XPASS tests/image/secret/test_sec_api.py::test_prime256v1_srk_table_parser
======================== 2500 passed, 21 skipped, 5 xfailed, 1 xpassed, 166 warnings in 443.63s (0:07:23) =========================
```

There are also some warnings printed by setuptools but these aren't critical and the the package still installs with them. This isn't really related to the issue in question but I thought I'd mention it.
```
 *      Package 'spsdk.data' is absent from the `packages` configuration.
 *      Package 'spsdk.data.common' is absent from the `packages` configuration.
 *      Package 'spsdk.data.common.crypto' is absent from the `packages` configuration.
 *      Package 'spsdk.data.common.memcfg' is absent from the `packages` configuration.
 *      Package 'spsdk.data.common.tp' is absent from the `packages` configuration.
 *      Package 'spsdk.data.common.xmcd' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.k32w1xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.kw45xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc550x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc551x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc552x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc553x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc55s0x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc55s1x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc55s2x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc55s3x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.lpc55s6x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mc56f81xxx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mcxa1xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mcxn23x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mcxn9xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mwct20d2x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mwct2xxxs' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mx8ulp' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mx93' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.mx95' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.nhs52sxx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rt101x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rt102x' is absent from the `packages` configuration.
[... cut ...]
 *      Package 'spsdk.data.devices.rt117x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rt118x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rt5xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rt6xx' is absent from the `packages` configuration.
 *      Package 'spsdk.data.devices.rw61x' is absent from the `packages` configuration.
 *      Package 'spsdk.data.jsonschemas' is absent from the `packages` configuration.
```

mstarecek, 2024-04-05:
Yes, we know about the setuptools warnings. It's not a big deal. Those folders contain only data files, no source code. 
I've tried to tackle those warnings but without any luck :( So we decided to leave it and don't burn time on something that actually works :D

closed 2024-04-05 by mstarecek