nxp-mcuxpresso/spsdk #67: lpsc55sxx secure boot: device starts also after erasing key store

jsiverskog, 2024-04-03:
hi,
after following the lpc55sxx_secure_boot.ipynb steps i can get my device to boot my master boot image (after some minor modifications).

however, as an experiment i tried to erase the key store:
```
%! blhost $CONNECTION set-property 29 1
%! blhost $CONNECTION write-memory 0x9E600 zero_1536.bin
%! blhost $CONNECTION set-property 29 0
```
after pressing the RESET button the device still starts up my binary.

is this expected?

jsiverskog, 2024-04-03:
i have tried to add root-of-trust config to the cmpa binary:
```
# Generate CMPA binary
%! pfr $VERBOSITY generate-binary -c $CMPA_TEMPLATE_PATH -o $CMPA_BIN -e $MBI_CONFIG_PATH --rot-config "workspace/cert_block_lpc55s6x.yml"
```
but i'm not sure whether that is the correct file or not. at least it boots, but it boots with erased key store also in this case.

mwsk, 2024-06-25:
In this case there is a signed image loaded in flash and it boots even if the key store is deleted. There is nothing strange about that, because for secure boot, no keys from key store are needed. For secure boot, it is only necessary that RKTH is entered in CMPA. The key store is only important for uploading SB files (SBKEK) and also if on-the-fly encryption internal flash using PRINCE is enabled.

closed 2024-06-25 by mwsk