nxp-mcuxpresso/spsdk #73: Errors testing 2.2.0: Invalid validation schema to check config: required must contain unique elements

jjakob, 2024-07-13:
I'm building spsdk 2.2.0 on Gentoo with my own ebuild, but am getting test errors (7 failed but they're all the same)

```
FAILED tests/nxpimage/test_nxpimage_fcb.py::test_nxpimage_fcb_export[rt5xx-flexspi_nor] - AssertionError: Expected code: 0, Actual
code: 1
SPSDK: Invalid validation schema to check config: required must contain unique elements
  File "/usr/lib/python3.12/site-packages/click/testing.py", line 408, in invoke
    return_value = cli.main(args=args or (), prog_name=prog_name, **extra)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/var/tmp/portage/dev-python/spsdk-2.2.0/work/spsdk-2.2.0/spsdk/apps/nxpimage.py", line 1887, in fcb_export_command
    fcb_export(config, output)
  File "/var/tmp/portage/dev-python/spsdk-2.2.0/work/spsdk-2.2.0/spsdk/apps/nxpimage.py", line 1898, in fcb_export
    check_config(config_data, schemas, search_paths=[os.path.dirname(config)])
  File "/var/tmp/portage/dev-python/spsdk-2.2.0/work/spsdk-2.2.0/spsdk/utils/schema_validator.py", line 235, in check_config
    raise SPSDKError(f"Invalid validation schema to check config: {str(exc)}") from exc
```

The same error is repeated for rt5xx-flexspi_nor, rt6xx-flexspi_nor, rt105x-flexspi_nor, rt106x-flexspi_nor, rt117x-flexspi_nor, lpc55s3x-flexspi_nor, mcxn9xx-flexspi_nor.

The ebuild I used is here https://github.com/jjakob/gentoo-overlay/blob/py3.12_staging_pynitrokey/dev-python/spsdk/spsdk-2.2.0.ebuild

BohdanMarek, 2024-07-15:
Hi @jjakob,
This problem is related to _fastjsonchschema_ package we use as a dependency.
The version 2.20.0 introduced additional check on validation schema where only the unique items are accepted in `required` keyword. For more information about this, see the [changelog](https://github.com/horejsek/python-fastjsonschema/blob/master/CHANGELOG.txt).

In terms of SPSDK, the current version 2.2.0 limits the fastjonschema to <2.20.0. See the [requirements.txt](https://github.com/nxp-mcuxpresso/spsdk/blob/master/requirements.txt#L11). In version 2.3.0, the validation schema will be fixed and the the upper boundary of _fastjsonchschema_ increased. Until then, please use the version lower then 2.20.0

closed 2024-07-15 by mwsk

jjakob, 2024-07-16:
Thanks, I opted to use your commit to patch the test, becase Gentoo only packages fastjsonschema 2.20.0 so downgrading is not an option.