nxp-mcuxpresso/spsdk #75: i.MX93 sign os container

yodakohl, 2024-09-25:
Is it possible to sign a os container for i.MX93 (linux + initramfs + dtb) using the spsdk?
If yes how does the yaml file have to look like?

marekvi95, 2024-09-30:
Hi,

As far as I know, only the U-Boot - two AHAB image container sets (u-boot, bl31, lpddrx, atf, ele firmware...) are authenticated by edge lock enclave. So only the u-boot must be signed. To build a signed u-boot refer to the documentation https://spsdk.readthedocs.io/en/latest/examples/imx93/imx93_signed_ahab_uboot.html
Just SRK table and signing key/signature provider needs to be specified.

In the upcoming SPSDK release 2.3, that will be released by the end of this week, we are adding command for signing the existing AHAB image/image container set without the need for parsing. We will also provide command for replacing the u-boot in existing WIC file (standard Yocto distribution).

I am referring to the standard BSP Yocto Linux distribution for SD/EMMC.  https://www.nxp.com/docs/en/user-guide/IMX_LINUX_USERS_GUIDE.pdf

yodakohl, 2024-09-30:
We are trying to switch from https://github.com/nxp-imx-support/meta-nxp-security-reference-design
that was using NXP CST.

This had the option to sing a container that is verified by u-boot using the auth_cntr command.

The code for the os-container signing is here: https://github.com/nxp-imx-support/meta-nxp-security-reference-design/blob/scarthgap-6.6.23-2.0.0/meta-secure-boot/recipes-secure-boot/linux/linux-imx-signature.bb

yodakohl, 2025-02-03:
Thank you, we succeeded by integrating a complete fit image in the AHAB container and specify the correct load address

closed 2025-02-03 by yodakohl