nxp-mcuxpresso/spsdk #82: nxpfuses: some CLI short flags are aliased/overloaded

devanlai, 2025-02-08:
Several `nxpfuses` commands inherit common nxpele CLI options to specify how to connect to the target device (via `@nxpele_options`), but then define new options with conflicting short flags.

I haven't every command across every tool, but here are the ones I noticed:

- In `getinfo`, `-d` for `--diff-only` conflicts with `-d` for `--device`
- In `print`, `-r` for `--rich` conflicts with `-r` for `--revision`

Where there's a conflict, the last-defined option seems to win, so `-d` for diff-only shadows `-d` for setting the device. This can be worked around by using the long flag names instead, but it would be best to either define an alternative short name or remove the short name altogether where there's a conflict.

BohdanMarek, 2025-05-12:
Hi @devanlai ,
Thank you for bringing this up. We already noticed this problem. It will be fixed in SPSDK 3.0.0

mwsk, 2025-05-28:
Hello  @devanlai, "--diff-only" was replaced with "-do" and --rich has only long flag name.

closed 2025-05-28 by mwsk