nxp-mcuxpresso/spsdk #83: support running programmatically

jsiverskog, 2025-03-27:
currently the `spsdk` package does not expose functions to do any actions programmatically / through an api.

we're executing `blhost` commands from another python program and it would be nice to not have to spawn subprocesses in order to do blhost actions.

it would be nice to be able to do something like this:
```
from spsdk import blhost

blhost.flash_image(..)
```

mstarecek, 2025-03-27:
Hello,
most of the time, using SPSDK via API is as simple as you suggested.
```python
with McuBoot(<interface>) as mboot:
    mboot.write_memory(<address>, <data>)
```

Granted, functionality such as `flash_image` is not that straight-forward. We're rolling out "twin functions" to CLI commands (see nxpdebugmbox -> `start_command` and `start` for example) So far adding this to blhost was not seen as a big priority.
So for now I'd suggest that you copy the content of `flash_image` function (perhaps ditching the progress bar?) or feel free to submit a PR?

jsiverskog, 2025-04-03:
@mstarecek : ok, thanks for the information.

i will see how i proceed :)

closed 2025-04-03 by jsiverskog